howerj/dbcc #4: dbcc  cannot process the byte order mixed package

ywwjzsf, 2018-05-30:
[VehCAN_GHAC.zip](https://github.com/howerj/dbcc/files/2052596/VehCAN_GHAC.zip)

hello:
howerj,
The attachment is VehCAN_GHAC.dbc .
I found a bug ,When i run the following command:
dbcc VehCAN_GHAC.dbc
It will generate tow files,VehCAN_GHAC.c and VehCAN_GHAC.h

then i found that a package with  byte order mixed can not be generate c code right.
first error:       <*data = reverse_byte_order(m)|>  c code has a syntax error,
second error: << shift operation is error when i process real can package, I can not get right result.

int pack_can_0x450_EVCM_cEM(can_0x450_EVCM_cEM_t *pack, uint64_t *data)
{
        register uint64_t x;
        register uint64_t m = 0;
        register uint64_t i = 0;
        /* ElecMachineSpeed: start-bit 7, length 16, endianess motorola, scaling 0.500000, offset -15000.000000 */
        x = (*(uint16_t*)(&pack->ElecMachineSpeed)) & 0xffff;
        x <<= 63;
        m |= x;
        /* ElecTorqueRequest: start-bit 31, length 13, endianess motorola, scaling 0.100000, offset -250.000000 */
        x = (*(uint16_t*)(&pack->ElecTorqueRequest)) & 0x1fff;
        x <<= 39;
        m |= x;
        /* AccelPedalPos: start-bit 23, length 8, endianess motorola, scaling 1.000000, offset 0.000000 */
        x = (*(uint8_t*)(&pack->AccelPedalPos)) & 0xff;
        x <<= 47;
        m |= x;
        /* BrakePedalDepressed: start-bit 32, length 1, endianess intel, scaling 1.000000, offset 0.000000 */
        x = (*(uint8_t*)(&pack->BrakePedalDepressed)) & 0x1;
        x <<= 32;
        i |= x;
        *data = reverse_byte_order(m)|;
        return 0;
}

note:My email is :ywwjzsf@126.com
you can send email to me to discuss this question.

howerj, 2018-05-30:
Hi, I'll look into it.

howerj, 2018-06-03:
Hi, I tried to email you, I got no response. I have fixed the compile error but the data conversion itself still looks a bit iffy - I have no way of testing things however.

ywwjzsf, 2018-06-06:
Thanks, i have a method to test this bug,
I have some can raw data, and have can-analyse PC software to analyse the can raw data with DBC file,
and then i will compare the result with your dbcc。
Wait for my news！

closed 2018-08-22 by howerj