howerj/dbcc #5: about decode and encode

chentyjpm, 2018-08-16:
wonderful work!

i have build it  in codeblock and its work well

translate dbc in c with helpful depack 

however 

i have find many encoder and decoder function with only  definition 

i may make it work with writing down my implementation? or maybe implementation in feature?

 Best regards

howerj, 2018-08-20:
Hi! Thanks! I'm not quite sure what you mean? Can you show me an example?

You can always work on your fork on a feature, and if it's sensible I can merge it back in. It might turn out that it's too specific too your situation.
Thanks again!

chentyjpm, 2018-08-22:
Hi! thanks for reply

for example

In DBC file

`BO_ 513 Bms_Cell_Limit_Volt_Info: 8 Vector__XXX
 SG_ BmsMaxCellVolt : 7|16@0+ (0.001,0) [0|65.535] "V"  TBOX_EV
 SG_ BmsMaxVoltCellNum : 23|8@0+ (1,0) [0|255] ""  TBOX_EV
 SG_ BmsMaxVoltModNum : 31|8@0+ (1,0) [0|255] ""  TBOX_EV
 SG_ BmsMinCellVolt : 39|16@0+ (0.001,0) [0|65.535] "V"  TBOX_EV
 SG_ BmsMinVoltCellNum : 55|8@0+ (1,0) [0|255] ""  TBOX_EV
 SG_ BmsMinVoltModNum : 63|8@0+ (1,0) [0|255] ""  TBOX_EV`

with this project "BmsMinCellVolt"  is convert to raw value in 0-65535 

but the actual value of "BmsMinCellVolt"  is 0.00 - 65.535 and end with "V" means 0.00V ~ 65.535V

this project can only decode in raw value 

i may convert it to actual value in my own code？
 
or there is some way to automatic convert it?
 
Best regards

howerj, 2018-08-22:
Hi!

There was a bug where the scaling and offset functions were not being generated, if you get the latest code you should see that it has been fixed. The functions are called 'encode' and 'decode'.

Let me know if this fixes your problem, 
Thanks,
Richard

chentyjpm, 2018-08-25:
hi 

thanks the problem has been fixed

^O^
Thanks again

closed 2018-08-25 by howerj