howerj/dbcc #7: encode function have err

chentyjpm, 2018-09-20:
1.  maybe encode function shoud input a double value and point of dst struct to process and output result true/false?

2.  value should first add offset then * scale now is first scale then offset  

for example

result is that

double decode_can_0x203_BattPackVoltage(can_0x203_Batt_Basic_Info_t *record)
{
	double rval = (double)(record->BattPackVoltage);
	rval *= 0.100000;
	rval += -1000.000000;
	return rval;
}

double encode_can_0x203_BattPackVoltage(can_0x203_Batt_Basic_Info_t *record)
{
	double rval = (double)(record->BattPackVoltage);
	rval *= 10.000000;
	rval += 1000.000000;
	return rval;
}

howerj, 2018-09-20:
Ah, wait, I see what you mean, I'll make functions that do the following:

bool encode(struct_t *record, double in);
bool decode(struct_t *record, double *out);
 
I'll also add range checks.

chentyjpm, 2018-09-25:
Hello!

first point is that

when decoder function you result is right

double decode_can_0x203_BattPackVoltage(can_0x203_Batt_Basic_Info_t *record, double *out)
{
double rval = (double)(record->BattPackVoltage);
rval *= SCALING;
rval += OFFSET;
*out = rval;
return (rval > MAXIMUM || rval < MINIMUM) ? false; true;
}

but when encode it should be that

double encode_can_0x203_BattPackVoltage(can_0x203_Batt_Basic_Info_t *record, double in)
{
double rval =in;
//first offset
rval -= OFFSET;
//scaling next
rval *= 1/SCALING;
record->BattPackVoltage = rval;
return (rval > MAXIMUM || rval < MINIMUM) ? false; true;
}

so the result is correct

Best wishs ^_^

howerj, 2018-10-21:
Can you check this works as expected? Then I will close this issue. Thanks!

chentyjpm, 2018-10-23:
checked  its right thanks ^_^