howerj/dbcc #9: C-Code generation multiplexing

xR3b0rn, 2018-10-12:
The signal multiplexing is  either leading to wrong code generation or the ast is build wrong.
I had not enought time to figured it out yet, a hint for me where useful so I may decide whether I hotfix it by my self and push it.

The wrong generated code looks like:
	register uint64_t x;
	register uint64_t i = (data);
	if(dlc < 8)
		return -1;
	x = i & 0xff;
	unpack->ID2 = x;
	switch(unpack->ID2) {
	case 12:
	x = (i >> 8) & 0xffff;
	unpack->ID_FN_RQ_BUS = x;
	break;
	case 12:
	x = (i >> 24) & 0xff;
	unpack->ID_RQ_BUS = x;
	break;
	case 12:
	x = (i >> 32) & 0xf;
	unpack->ST_RQ_BUS = x;
	break;
	default: return -1;
	}
	return 0;

howerj, 2018-10-12:
Hi! Thanks for the report, could you post the signal and multiplexor in the dbc file that generates this code? I can reproduce this bug, but only if I make my the input file invalid (well, invalid for conversion).

I have an example of a multiplexor in 'ex1.c', but only one. If you go to line 696 of 'ex1.dbc', there is a signal with a multiplexor. 

```
BO_ 666 IMU5: 4 Vector__XXX
 SG_ multiplexor M:   0|4@1+ (1,0) [0|15] "multiplexor" Vector__XXX
 SG_ multi1 m1:  8|4@1+ (1,0) [0|15] "multiplexed-a" Vector__XXX
 SG_ multi2 m0:  8|4@1+ (1,0) [0|15] "multiplexed-b" Vector__XXX
 SG_ multi3 m2:  8|4@1+ (1,0) [0|15] "multiplexed-c" Vector__XXX
 SG_ multi4 m3:  8|4@1+ (1,0) [0|15] "multiplexed-d" Vector__XXX
 SG_ normal:  16|8@1- (1,0) [-128|127] "normal" Vector__XXX

```

The case labels are generated from the 'm0...m3' values. If these are the same, then the case labels will be the same. This is not ideal as it leads to compiler errors. I think dbcc should detect this, and issue an error. This could be checked in the 'multiplexor_switch' function, line 294 of '2c.c'.

xR3b0rn, 2018-10-13:
I am atm not able to send an minimal example of my error. But guess, you are taking the casees from m1, m0, and so on. In my dbc File I have multiply m12s. So is the error coming from this?

On this site I found an example with multiply m-cases too:
http://socialledge.com/sjsu/index.php/DBC_Format

xR3b0rn, 2018-10-13:
I did't this for you:
```
static int multiplexor_switch(can_msg_t *msg, signal_t *multiplexor, FILE *c, bool serialize)
{
	int cmp(const void* lhs, const void* rhs)
	{
		int ret = 0;
		if ((*(signal_t**)lhs)->switchval < ((*(signal_t**)rhs)->switchval))
		{
			ret = -1;
		}
		else if ((*(signal_t**)lhs)->switchval > (*(signal_t**)rhs)->switchval)
		{
			ret = 1;
		}
		return ret;
	}
	fprintf(c, "\tswitch(%s->%s) {\n", serialize ? "pack" : "unpack", multiplexor->name);
	qsort(msg->signals, msg->signal_count, sizeof(*msg->signals), cmp);
	for(size_t i = 0; i < msg->signal_count; i++)
	{
		signal_t *sig = msg->signals[i];
		if(!(sig->is_multiplexed))
			continue;
		fprintf(c, "\tcase %u:\n", sig->switchval);
		size_t j = i;
		for (; j < msg->signal_count; j++)
		{
			if (msg->signals[i]->switchval != msg->signals[j]->switchval)
			{
				break;
			}
			signal_t* sig = msg->signals[j];
			if((serialize ? signal2serializer(sig, c) : signal2deserializer(sig, c)) < 0)
			{
				return -1;
			}
		}
		i = j;
		fprintf(c, "\tbreak;\n");
	}
	fprintf(c, "\tdefault: return -1;\n\t}\n");
	return 0;
}
```
Btw., am I able to push such code?

xR3b0rn, 2018-10-13:
I created an pull request.

howerj, 2018-10-13:
I've also pushed a fix, your code looks better, so I'll merge that. However, it does not appear to generate correct code under all circumstances. I have missing case statements when I pass it the example message 'BO_ 666 IMU5: 4 Vector__XXX' I've added to ex1.dbc.

Also before merging could you move the nested function outside of 'multiplexor_switch' and change the brace style so it matches the rest of the project?

Thanks for your work!

closed 2018-10-13 by howerj