howerj/dbcc #13: Support for DBC float and double types?

karlding, 2018-11-13:
Hi,

I was wondering if you were planning on adding support for `float` and `double` types?

`dbcc` currently treats `float` and `double` types as a signed value of the same size (ie. `int{32,64}_t`) when generating code, which results in incorrect values being encoded/decoded.

From a quick scan through the code, it seems like you already have an attribute on the `struct` allocated for floating-point types (`is_floating`), and the codegen code accounts for it, but it just isn't getting set anywhere when parsing the DBC.

This is indicated via the `SIG_VALTYPE_` attribute in the DBC.

A `float` value is denoted via 

``` 
SIG_VALTYPE_ <can-id> <signame> : 1;
``` 

And a `double` is denoted via

``` 
SIG_VALTYPE_ <can-id> <signame> : 2;
```

Thanks!

howerj, 2018-11-13:
Hi!

I didn't have any plans to, because I didn't have an example of a DBC file that encoded floats, it's not something that I've really needed. It can be added fairly easily now that you've given an example of one.

Could you give a full example of a minimal file I can test this on? One that encodes a single and a double?

Also, in your example, is the CAN ID the numerical value or the message name?

Thanks,
Richard.

karlding, 2018-11-14:
Sure!

Here's [2 DBC files](https://github.com/howerj/dbcc/files/2579255/signals.zip) that I've generated using [Kvaser Database Editor](https://www.kvaser.com/downloads-kvaser/).

The CAN ID is the numerical value (ie. the arbitration ID).

howerj, 2018-11-14:
Cheers! I'll let you know I've gotten around to implementing this, it shouldn't take too long.

howerj, 2018-11-15:
Ok, I've added floating point support, no doubt there are problems and I have not tested the functionality (unit tests are on my to do list).

The packing/unpacking code does not handle all case either (Inf and NaN are not decoded or encoded). I should also add an option to allow type punning for conversion instead of using the unpack754/pack754 routines.

closed 2018-11-17 by howerj