howerj/dbcc #19: Plans for value lists?

martonmiklos, 2018-11-21:
Hi Richard,

I have seen this comment about the value list support:
https://github.com/howerj/dbcc/blob/master/parse.c#L85

What is your plan about the value list support?
I am thinking about generating an enum from these lists would be useful as well setting the value mapped fields to enum in the generated code.

If you have other plans please let me know, because I think I might be able to came up with an implementation.

howerj, 2018-11-21:
Hi Marton,

If you want to come up with an implementation, feel free to do so! I do not have any plans to do anything with them at the moment (I do not even know what they look like in a DBC file). I mostly implement things because other people want them, so if you get stuck or need pointers I can help. 

Thanks,
Richard

martonmiklos, 2018-11-21:
Well I am new to the whole DBC area, but for me (by peeking with the Vector's DBC editor) it looks that they are kind of similar to the enums in C. As a first step I will get them parsed and assigned to the appropriate signals and I will generate enums to the headers. Later we can discuss if we want to use these enums in the pack/unpack/print methods.

howerj, 2018-11-21:
Ah, I meant I do not know what the specific fields look like, I used to DBC examples that contained examples of 'VAL_TABLE_', I do not any more. But that's good, let me know when you want to talk about how pack... and other methods should look. I also need to think about the other file formats as well (but they're not as important as the C generation).

closed 2019-02-24 by howerj