howerj/dbcc #21: Too small offsets and scales generate wrong code

badicsalex, 2018-11-27:
We have a CAN DBC, where scale is 1E-7. The code generated is something like
`value *= 0.000000;`
Which is obviously wrong. It should be
`value *= 1e-7;`
It is caused by using "%f" (always use decimal notation) instead of "%g" (always use the shortest notation) in the printf calls all over the code.

I can easily prepare a pull request where every "%f" is replaced with "%g" in the code base, if you want.

howerj, 2018-11-27:
Feel free to make a pull request, I'll test it on all platforms. The C library on Windows is sometimes a bit funny with certain format specifiers (even the standard ones), I'll have to check it is supported. There's also '%.*f', or just using '%e'?

badicsalex, 2018-11-27:
First I tried %e, which works great, but %g generated somewhat prettier code.

closed 2018-11-27 by howerj

badicsalex, 2018-11-27:
Thanks for such a fast response :)