howerj/dbcc #30: Make NS_ and BS_ parser syntax slightly more flexible

nkirkby, 2019-05-15:
Hi, thanks for this fantastic library!
I had to tweak the parser's regular expressions to get it to munch a DBC I came across that uses spaces instead of a tab before `NS_`, as well as an extra space in `BS_` lines:
https://github.com/nkirkby/dbcc/commit/5247f2a7bbb44d386c03764eab1950f3a2a5854d

I haven't been able to dig up a DBC specification, so I'm unsure if the DBC I'm parsing is incorrectly formatted.

howerj, 2019-05-15:
Cool, just pushed a fix, which is what you put in your fork. Could you make a pull request next time?

Cheers for that! Let me know if there are any problems.

closed 2019-05-15 by howerj

nkirkby, 2019-05-15:
> Cool, just pushed a fix, which is what you put in your fork. Could you make a pull request next time?
certainly.  thank you