howerj/dbcc #31: suggest:  about decode param type

chentyjpm, 2019-05-30:
Thanks for the work~~

i have a problem

when i use the latest version, it convent different type of value in different types which cause me a little problem.

When use Code C I can not detect the value type in a function param

so when use code generate .... ... this way of coding is failed

`
#define DEC_MOD(obj,srcid,matchid,fun,ID) \
if(srcid == matchid)\
{\
	double tmpval = 0; \
	fun(&obj, &tmpval); \
	setval(report, ID, tmpval); \
}

...
	if (unpack_message(&canobj, canid, candat, 8) == 0)
	{
DEC_MOD(&canobj, canid, 0x9002fff4, decode_can_0x9002fff4_BAT_CU, GBID_FC_VELCUR)
 // 0x9002fff4.BAT_CU->GBID_FC_VELCUR
DEC_MOD(&canobj, canid, 0x9002fff4, decode_can_0x9002fff4_BAT_VOL, GBID_FC_VELVOL)
 // 0x9002fff4.BAT_VOL->GBID_FC_VELVOL
...
`

maybe there can be a switch to change decode function param to all is in double like this?
or switch decode function like this

`
//current
int encode_can_0x9827fff7_TEM_SEN_23(can_obj_can_h_t *o, double in);
int decode_can_0x98eaff00_PGN(const can_obj_can_h_t *o, uint32_t *out);

//add paramtype in function name
int encode_double_can_0x9827fff7_TEM_SEN_23(can_obj_can_h_t *o, double in);
int decode_uint32_can_0x98eaff00_PGN(const can_obj_can_h_t *o, uint32_t *out);

//or all param to double
int encode_can_0x9827fff7_TEM_SEN_23(can_obj_can_h_t *o, double in);
int decode_can_0x98eaff00_PGN(const can_obj_can_h_t *o, double *out);
`
so i can use script to detect the function param type and process

howerj, 2019-05-30:
Hi, that's a good suggestion, I have pushed a commit which adds a command line option that does what you want - if you use the '-D' option then only doubles are used for the encode/decode functions. 

Let me know if that is what you want.

For you second suggestion, adding the parameter type to the function name, the parameter type is already there, it might just be a little bit more work to parse out. Check out the perl script 'extract.pl' I have added in the 'out/' directory of the project. Of course you could even modify the program to put the function/type information into a nice CSV file if you wanted.

howerj, 2019-06-01:
I'll mark this as closed now, thanks.

closed 2019-06-01 by howerj

chentyjpm, 2019-06-02:
thanks nice work~~~