howerj/dbcc #32: cannot decode messages that share same address

sume-cn, 2019-08-26:
for example:
 SG_ Speed_ms : 39|16@0+ (0.00514444,0) [0|0] "m/s" Vector__XXX
 SG_ Speed_mph : 39|16@0+ (0.0115078,0) [0|0] "mph" Vector__XXX
 SG_ Speed_kmh : 39|16@0+ (0.01852,0) [0|0] "km/h" Vector__XXX

it will report:
warning: Too many signals, not enough bytes (DLC is too low, fix your DBC file)

howerj, 2019-08-26:
I've added those lines to a message in the example DBC file and I believe it generates the code that you have specified correctly (I haven't actually run it), all it does is emit a warning. Can you give me more information on the incorrect code it is generating? What do you see when you run the code generator and what do you expect?

sume-cn, 2019-08-27:
Sorry, my bad. All codes was generated successfully. I only noticed the warning.

closed 2019-08-27 by sume-cn