howerj/dbcc #35: Decoding float signal: wrong type?

npnmos, 2021-07-29:
Hello, 
Nice tool! I have just noticed a strange thing, that i do not know if it is intended...
When I do `dbcc float_signal.dbc` I get a function like this:

 ```
int decode_can_0x400_FloatSignal0(const can_obj_float_signal_h_t *o, int32_t *out) {
     assert(o);
     assert(out);
     int32_t rval = (int32_t)(o->can_0x400_NewMessage0.FloatSignal0);
     *out = rval;
     return 0;
 }
```

Why should it return a `int32_t`? Why not returning a `float`?

howerj, 2021-07-29:
You're correct, I've added a fix. Let me know if you have other problems with the floating point encoding/decoding as it is probably the least tested bit of the generated code.

closed 2021-07-29 by howerj

npnmos, 2021-07-30:
Nice! I saw you fix but did not manage to test it yet.
However, i think the same happens for 1 bit flags: they are managed as uint8_t, while they might be treated as boolean

howerj, 2021-08-04:
That's deliberate, just because it is a 1 bit flag, does not mean true/false is applicable.