howerj/dbcc #36: Missing infromation about CAN-FD on main page

AdrianChemicz, 2021-08-10:
Hi,

Great code. Idea of this tool is perfectly for me but unfortunatelly CAN-FD is not supported.
For my and probably not only mine purpose it cause that this tool is useles :(
Number of users which will have this problem will grow because CAN-FD enter to more places. In automotive currently most new projects which I know use CAN-FD.

Please add on main page information that CAN-FD isn't supported.

Result after call application with CAN-FD database:
D:\dbc>dbcc.exe CAN_FD_XXXXXXXXX.dbc
Assertion failed!

Program: D:\dbc\dbcc.exe
File: ../can.c, Line 130

Expression: r == 1 && sig->start_bit <= 64

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.

I will try solve it on my own(with succes or not). If I will create something usefull I will inform you.

howerj, 2021-08-10:
Hello!

That's understandable, although I do not have much interest at the moment in doing major development of the project as I no longer work in the automotive industry and have not for a while. I also do not have access to CAN-FD specifications or files for testing and development.

I shall update the project with a big notice about CAN-FD.

I should keep that assertion in there, although the assertion should not fire and it should be detected and an error returned instead, which is a minor bug.

Let me know if you manage to do something useful, I am happy to merge in code back into master. I suspect the best thing to do would be to make a new "2c.c" file for CAN-FD as a lot of the code generation in that file assumes the maximum message size is 8-bytes and is kind of tied to that (as everything can fit nicely into a uint64\_t).

Thanks! And I will keep this open for now.

AdrianChemicz, 2021-09-09:
Current status is that I need my own HW for CAN-FD. I thought to order my favourite LCP microcontroller but unfortunately we have chip crisis so I change approach and bought board with SPI to CAN-FD chip.

I gain knowledge about HW and I send to factory order to create first prototype of board.

When I have HW environment then I will try to somehow provide support for CAN-FD to your code and share with you.

I aware you that I'm not fast and It can take time.

howerj, 2021-09-10:
No problem! I'm not in a rush. I like to close issues that I am not working on to keep things tidy though. If you get around to implementing CAN-FD, or to ask questions about it, open up a new issue or make a pull request.

I'm more than willing to accept new changes.

closed 2021-09-10 by howerj