howerj/dbcc #38: Can't get CM_ string when it has \".

takaite, 2022-09-26:
Hello;

could you help me with a small issue? I'm trying to parse a dbc file that has the **\\"** scape sequence on comments. It's making parse to stop to process with no error, but all the following fields are ignored.

I've tried to change the regex string but could not reach a satisfactory solution.

A example of string to parse:
`CM_ SG_ 2628452094 MF_SourcePathNameLength "Length of \"source\" path name string for Move File";`

howerj, 2022-09-27:
This will require a short grammar change as the string type currently does not deal with escape characters, I don't know when I'll get around to it, I have a new born baby.

howerj, 2022-11-17:
This is partially fixed

takaite, 2022-11-18:
Oh good! I'll give a try latter. Tks by the moment!

howerj, 2023-03-08:
Did it work?

takaite, 2023-03-14:
Hi James;

Sorry for the delay in answering, but I was in another project without time to test.

I didn't test all the release because I'm using just some parts of your code, but the new pattern work for me.
Just as note: I did not check if the comment content is ok because I don't care about it, but it is not breaking the parse, going until the the end of the DBC.

closed 2023-03-14 by takaite