howerj/dbcc #39: Best way to get message length

warrickw, 2023-02-08:
I'm using `sizeof(can_0x003_trim_manual_t)` to get the frame size to write. Is this the best way to do this? 

I wondered if the `pack_message` function could return the length of the relevant message, rather than just 0 for success... but obviously that's a breaking change. If doing a `sizeof` the relevant message struct is the best approach then I'll just keep on with that.

howerj, 2023-02-08:
That's not the best way of doing it, or more accurately, it's currently the only way of doing it, but it's not good. It *may* give the correct result, but it is depedent on the compiler and platform the code is compiled on and the padding it decides to insert into the structure.

Although a breaking change, it would be a good one, it should not be too difficult to add.

howerj, 2023-02-08:
Changed it to do what you said, that should work now.

closed 2023-02-08 by howerj

warrickw, 2023-02-08:
Looks good and working well for me. You certainly implemented that faster than I could have! Very useful tool, I'm using it to generate the CAN structures for systems in an experimental airplane built on the STM32 platform.

howerj, 2023-02-09:
This one was easy to implement, it's always nice to know what people are using this tool for! Cheers!