howerj/dbcc #44: Long signal value generates invalid C code.

FZGabriel, 2023-03-24:
The code generated from this following signal value is broken:

```dbc
VAL_ 2147483938 IVT_SleepAck 15 "reserved 15" 14 "reserved 14" 13 "reserved 13" 12 "reserved 12" 11 "reserved 11" 10 "reserved 10" 9 "reserved 9" 8 "reserved 8" 7 "reserved 7" 6 "reserved 6" 5 "reserved 5" 4 "reserved 4" 3 "reserved 3" 2 "reserved 2" 1 "Sleep request acknowledged" 0 "Sleep request not acknowledged" ;
```

The broken code.
```C
typedef enum {
	IVT_SleepAck_Sleep request not acknowledged_e = 0, //< Look here
	IVT_SleepAck_Sleep request acknowledged_e = 1,      //< Look here
	IVT_SleepAck_reserved 2_e = 2,
	IVT_SleepAck_reserved 3_e = 3,
	IVT_SleepAck_reserved 4_e = 4,
	IVT_SleepAck_reserved 5_e = 5,
	IVT_SleepAck_reserved 6_e = 6,
	IVT_SleepAck_reserved 7_e = 7,
	IVT_SleepAck_reserved 8_e = 8,
	IVT_SleepAck_reserved 9_e = 9,
	IVT_SleepAck_reserved 10_e = 10,
	IVT_SleepAck_reserved 11_e = 11,
	IVT_SleepAck_reserved 12_e = 12,
	IVT_SleepAck_reserved 13_e = 13,
	IVT_SleepAck_reserved 14_e = 14,
	IVT_SleepAck_reserved 15_e = 15,
} IVT_SleepAck_e;

```

Thanks for the tool. 
Compiling and using it was painless :)

FZGabriel, 2023-03-24:
Additional comments:
I just noticed that signal values with the `+` and `-` characters are not handled properly.

```dbc
VAL_ 2147483920 IVT_Ctrl_Fuse_State 1 "Fuse+ AND Fuse- closed" 0 "Fuse+ OR Fuse- open" ;
```
The generated code:
```C
typedef enum {
	IVT_Ctrl_Fuse_State_Fuse+ OR Fuse- open_e = 0,
	IVT_Ctrl_Fuse_State_Fuse+ AND Fuse- closed_e = 1,
} IVT_Ctrl_Fuse_State_e;
```

howerj, 2023-03-25:
Looks like a problem! It should be a fairly easy fix though.

howerj, 2023-03-25:
Do you have an example DBC file with these in?

closed 2023-07-24 by howerj