howerj/dbcc #45: Generated code do not support extended messages

FZGabriel, 2023-03-24:


howerj, 2023-03-25:
Ahoy! The CAN ID 2147483904 is 0x80000100, I might be wrong, but that seems invalid? (unless that top bit is set for other reasons)
Cheers,
Richard

FZGabriel, 2023-03-27:
I think the top bit is used in the `dbc` format to indicate that the CAN ID is in Extended mode.

FZGabriel, 2023-03-27:
I think the top bit is used in the `dbc` format to indicate that the CAN ID is in Extended mode.

Here's a test to prove my hypothesis:
[image]

howerj, 2023-03-31:
I'm working on a branch called 'fixes-and-reorg', which should address the issues you've raised. I think it has fixed all of them, just need to do some testing and more work.

closed 2023-07-24 by howerj